tanaes/whopping_Voron_mods #33: LED Always ON

hreality, 2023-10-20:
Verified my logic and the probe is functioning... Voron 2.4 r1 Octopus 1.1 never needed a diode and ran klicky standard without issue.  Thing is, the LED is always on dims briefly while probing..

[probe]
##  Inductive Probe - DIAG7 Replaced with Klicky PCB
##  This probe is not used for Z height, only Quad Gantry Leveling
pin: ~PB7
x_offset: 0
y_offset: 19.75
z_offset: 6.42
speed: 4
samples: 3
samples_result: median
sample_retract_dist: 3.0
samples_tolerance: 0.3
samples_tolerance_retries: 4

karlfife, 2023-12-23:
Same issue.  LED is on even when probe is NOT attached. 
Not sure how the circuit is built, but tried configuring signal to pull up, and pull down.  Same result

tanaes, 2023-12-23:
I recently got a batch of PCBs from JLCPCB in which half the boards were assembled with the wrong LED part number, causing the same behavior. I wouldn't be surprised if other vendors received boards with the same problem.

karlfife, 2023-12-23:
I must misunderstand something about the theory of operation of the PCB Klicky.  Since the magnetic probe itself has both NO and NC contacts, I assumed the only electrical path to illuminate the status LED was via Normally Closed contact on the probe.   i.e. No probe, no path to ground for the LED. When the probe is attached, current flows via the NC terminal through the diagnostic LED to ground, causing the LED to illuminate. 
When the attached probe is triggered, the NC circuit opens, interrupting connection to the LED, instead pulling down 'signal' to ground.  
I must have this wrong, otherwise, even the wrong type of LED could not stay lit.  What is the design if not as above?   Maybe the circuit is totally different, with an LED zener diode with the wrong breakdown voltage, and that's what causing it to remain lit?  Care to shed light?

tanaes, 2023-12-23:
Ahh sorry i thought you were talking about OptoTap! Please ignore my
comment then. 😅

—
<https://github.com/tanaes/whopping_Voron_mods/issues/33#issuecomment-
auth/AB7ISAGC4XDGY6TMUP3FU4TYK43KBAVCNFSM6AAAAAA6IARXWWVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTQNRYGM3DEOJTHE>.

Rudis1261, 2025-04-07:
I also had this issue, I double checked my board (Octopus) pinout and I had the Signal and GND pins flipped. After flipping them works beautiful 👌