tanaes/whopping_Voron_mods #40: Clicky Clacky Latch pivots within the extrusion as your tighten it down

rnbarrett, 2024-03-01:
This pivoting pulls the leading edge of the latch away from the frame, causing it to interfere with proper handle movement.

Could you consider adding a "spacer" to fill the gap between the roll-in nut and the bottom of the latch?  I think it will stop the pivoting without having to do something like make the latch project deeper (or wider) into the extrusion, which would likely mean you'd need different latches for dealing with the known variances between manufacturers.

Something like this (from the A/B drive mounts on a Voron 2.4R2) is what I'm talking about.
[image]

Or are we really supposed to be using hammerheads?  :)

rnbarrett, 2024-03-01:
Perhaps something like this (the red areas) - leaving the M3 holes of course.
[image]
[image]

rnbarrett, 2024-03-01:
Video that I think shows what I'm talking about with the pivot
https://github.com/tanaes/whopping_Voron_mods/assets/31315019/c1fa6dd1-d0e1-4a6b-93b4-990f4beb145e

tanaes, 2024-03-04:
This is a real challenge for me. I actually had a projection very much like that in initial testing. The problem I encountered was that it limited the force with which the hinges could clamp onto the extrusion -- with hammerhead nuts that had a larger projection there, you could tighten it right down and the hinges would slowly slide under the door weight over time.

All of this boils down to needing to accommodate panels that extend to the very edge of the extrusion slot, so there's no room for balanced engagement across both sides of the extrusion.

I'm open to ideas!