tanaes/whopping_Voron_mods #42: Hinges

Macmagoo, 2024-04-22:
I have a LDO Trident 300 cubed. I bought the extrusions for the Clicky Clack door mod but when I attach the door to the hinges it is half an inch or so away from the frame at the latch side. Any advice appreciated. Thank you.

miklas206, 2024-07-25:
I have the same i dont know if its just how it is or if its a problem plzz let us know :)

southproduct, 2025-06-14:
[image]

Hi, I'm not the developer, but I will try to help.

Make sure the hinge system is as depicted. If so, the gap should be minimal (1-4 mm or 0.0393701" - 0.15748") Jesus Christ! I can't stand imperial measurements.

1/2 an inch anywhere sounds unrealistic. Try to attack pictures of the issue (with the door attached / un-attached) make sure to take pictures of all the parts as well. It's also helpful to have a ruler or a cutting mat with a printed ruler on the background for reference. That way anyone can see if the printed parts are way off measurement.

Hope this helps.

DocFoFo, 2025-06-15:
I'm with SouthProduct on this one. Sounds like you have the hinges mixed up. Everything worked out well for me. There is supposed to be a gap, but it's for the seal. The frame should not be up against the printer. There is foam that goes between the door printer.