tanaes/whopping_Voron_mods #46: Clicky Clacky There is a narrow gap when the door is closed.

skyaee, 2024-08-01:
Hello. No matter how I install and adjust it, the door cannot be completely sealed when closed. There is still a narrow gap. This troubles me. Is there any solution?

martinewald, 2024-12-30:
Stupid question, did you add the foam tape on the inside of the door?

southproduct, 2025-06-14:
I'm not the designer. I couldn't import the .f3d source files, but I managed to import the provided .STL files and added it to my custom Voron.

In the real world, after applying the foam tape inside, I can see there's a ~1mm gap near the corner and a ~4mm gap near the latching system.

[image]

In my CAD app, I can see that if I place the door frame parallel to the frame of the printer there would be a 3mm gap.

---

I've requested the developer to include .STEP files, hopefully it includes the entire frame (+foam tape) that way we can see how the design is intended to work.

Regardless, if they don't... you can always add thicker and thicker tape until it works.

For example, in my case I had the wrong printer settings and the holes for the hinge system were wrong... I didn't reprint. I Macgyver'ed a solution with playdough and the latch works perfectly fine.

Hope this helps. 

Cheers!